wellenvogel/avnav #414: Zoom: Higher zoom level tiles shown instead of available lower level tiles

ronangaillard, 2025-04-26:
(Sorry it's me again)

Sometimes shown tile are from a upper zoom level even if a lower zoomed level tile is available. Sorry if this is not clear, here is a video showing the issue :

https://youtu.be/-LFwuQlypfQ

Mbtiles used is the same as in the other issue : https://drive.proton.me/urls/X4VR2QKCH8#0RIGNu3CVM5c

Here is the issue shown in images :

**Before zoom**

[image]

**During zoom**

[image]

**Zooming more**

[image]

Lower part is using a tile from an upper level but a lower tile is available.

wellenvogel, 2025-04-26:
Are you sure about "upper" and "lower"?
In your screenshot the lower tile is from a lower level (i.e. scaled up).

Btw: Would be helpful to describe your system setup including the versions of AvNav (and related software).

ronangaillard, 2025-04-27:
After further investigation it's in fact showing the bottom (underneath) chart layer on the bottom side of the last screenshot. Meaning that if I remove the second chart layer I only see a white space :

[image]

Why would some tile disappear at some zoom level but others don't ? I also think so they should always be visible even if overzoomed. We often need to zoom a lot on tiny screens.

You are right about system setup :

- raspberry pi 5 with 1920x1080 hdmi display
- raspbian bookworm official image 
- avnav webui in firefox in kiosk mode
- avnav version 20240616

I'm also able to reproduce the issue with a docker image on my MacBook 

```

FROM --platform=linux/arm64 debian:bullseye

RUN apt-get update && apt-get install -y --no-install-recommends \
         wget \
         gnupg \
         ca-certificates \
         && rm -rf /var/lib/apt/lists/*

RUN wget -O - https://www.free-x.de/debian/oss.boating.gpg.key | gpg --dearmor | tee /usr/share/keyrings/oss.boating.gpg
RUN echo "deb [signed-by=/usr/share/keyrings/oss.boating.gpg] https://www.free-x.de/debian bullseye main contrib non-free" | tee /etc/apt/sources.list.d/boating.list

RUN apt-get update && apt-get install -y --no-install-recommends \
         avnav \
         avnav-ocharts-plugin \
         avnav-oesenc \
         avnav-update-plugin \
         && rm -rf /var/lib/apt/lists/*

EXPOSE 8080

VOLUME /home/avnav/avnav

RUN mkdir -p /home/avnav/avnav && chown avnav:avnav /home/avnav/avnav

USER avnav

CMD ["/usr/bin/avnav"]
```

The issues is the same then in firefox or chrome.

wellenvogel, 2025-04-27:
I will just check with your mbtiles file...
In general AvNav will by default just load the tiles that match the zoom level. During zooming OpenLayers will potentially zoom out existing tiles - but load new ones when done. Most probably there are no tiles at the requested zoom level - so loading the tile will simply fail.
With "zoom up lower layers" AvNav will just try to load lower zoom level tiles if the requested tile is not there. But this is limited to 4 levels by default as it costs additional network traffic. The viewer part in the browser does not really know about the available tiles...

wellenvogel, 2025-05-04:
I investigated a bit further using your mbtiles file.
This file contains tiles that are completely transparent - example: 15/16192/11280.png.
AvNav has no idea if the content of the tile is correct - it simply shows the tile that is there.
So I guess you should ensure that the mbtiles file does not contain invalid transparent tiles. Not sure how you created them - but maybe the source layer did not have data at that zoom level and you added transparent tiles.

ronangaillard, 2025-05-04:
Thanks for that investigation !

I created them using Mobile Atlas Creator, not sure why it created such tiles.

I'll look into options, but I guess we can close this issue, thanks !

closed 2025-05-04 by wellenvogel