wellenvogel/avnav #416: Hide or Reorder User Apps

ronangaillard, 2025-04-27:
Sorry it's me again...

Hi, the SignalK User App is first in my list, but I don't use SignalK, so it just shows a connection error.

My Pypilot app is second, so I always have to click past the useless SignalK page to get to Pypilot. This extra click is inconvenient on a boat.

Do you think we could have a way to either:

- Hide User Apps we don't use?
- Or change the order of User Apps so we can put the most important one first?

Or is there a solution I did not see ?

Thanks!

wellenvogel, 2025-04-27:
If you don't use SignalK just disable the SignalK handler. This way the SignalK button will go away.

ronangaillard, 2025-04-27:
Thanks, it does indeed work when changing the setting through the gui (I thought I could disable it using the server xml config like `<AVNSignalKHandler enabled="false"/>` but it did not work). Doing it using the GUI works ok 👍

closed 2025-04-27 by ronangaillard