wellenvogel/avnav #417: Configuration Waypoint Alarm Android

PietKnot, 2025-04-28:
when sailing a route on inland waterways sometimes with a lot of waypoints, the waypoint alarms become disturbing: 3x sound-riddle and red screen-popup that must be confirmed. Is there a way to disable this screen popup and possibly reduce the number of number sounds-riddle (once would be enough for me without screen popup) or switch off completely.

Webapp has option: prevent-alarms
AVNAlarmHandler configuration available on Android somehow (did not see one)?

Would be nice to be able to do so.

wellenvogel, 2025-04-28:
In the android system settings (android button - only available in the app directly, not in connected browser) you can disable (or change) the alarm sounds.
Just select a sound that fit's your needs.
I guess currently there is no chance to disable alarms completely.

PietKnot, 2025-04-28:
Sorry, missing that one; did see it in the past to set ExternalStorage but never paid attention to the seconde setting. Good enough, thanks for the hint.

closed 2025-04-29 by PietKnot