wellenvogel/avnav #421: Cannot install new ocharts

Fietsenwiel, 2025-05-21:
I have now working ocharts but they are from 2022.
So now I wanted to install new ocharts but this goes wrong (this is the provider.log):

`2025/05/21-13:09:04.403-0x7f69bf6c00-INFO-creating chart File /home/pi/avnav/data/ocharts/charts/oeuSENC-UK-2025-1-2-base-RelexedPi/OC-44-991203.oesu
2025/05/21-13:09:04.420-0x7f69bf6c00-INFO-creating chart File /home/pi/avnav/data/ocharts/charts/oeuSENC-UK-2025-1-2-base-RelexedPi/OC-44-E28704.oesu
2025/05/21-13:09:04.462-0x7f69bf6c00-INFO-creating chart File /home/pi/avnav/data/ocharts/charts/oeuSENC-UK-2025-1-2-base-RelexedPi/OC-44-D20805.oesu
2025/05/21-13:09:04.474-0x7f69bf6c00-INFO-creating chart File /home/pi/avnav/data/ocharts/charts/oeuSENC-UK-2025-1-2-base-RelexedPi/OC-44-D12805.oesu
2025/05/21-13:09:04.666-0x7f69bf6c00-INFO-found chart OC-44-571003.oesu to try, create temp dir with it
2025/05/21-13:09:04.682-0x7fa5c5a7c0-INFO-TryMessage::Process
2025/05/21-13:09:04.682-0x7fa5c5a7c0-INFO-ChartManager: TryOpenChart /home/pi/avnav/data/ocharts/charts/__tmp2025-05-21-13-09-04-667222/OC-44-571003.oesu
2025/05/21-13:09:04.683-0x7fa5c5a7c0-INFO-opening /home/pi/avnav/data/ocharts/charts/__tmp2025-05-21-13-09-04-667222/OC-44-571003.oesu
2025/05/21-13:09:04.700-0x7fa5c5a7c0-ERROR-File '/home/pi/avnav/data/ocharts/charts/__tmp2025-05-21-13-09-04-667222/Chartinfo.txt' couldn't be removed (error 2: No such file or directory)
2025/05/21-13:09:04.701-0x7fa5c5a7c0-INFO-ShowEULA
2025/05/21-13:09:04.710-0x7fa5c5a7c0-INFO-EULA Accepted.
2025/05/21-13:09:04.711-0x7fa5c5a7c0-INFO-processChartInfo adding config_string: British Isles 2025;2025/1-2;2026-05-14
2025/05/21-13:09:04.711-0x7fa5c5a7c0-INFO-Computed installBase: 
2025/05/21-13:09:04.711-0x7fa5c5a7c0-INFO-Loading keyFile: /home/pi/avnav/data/ocharts/charts/__tmp2025-05-21-13-09-04-667222/oeuSENC-UK-2025-1-2-base-RelexedPi.XML
2025/05/21-13:09:04.735-0x7fa5c5a7c0-INFO-Loading keyFile: /home/pi/avnav/data/ocharts/charts/__tmp2025-05-21-13-09-04-667222/ChartList.XML
2025/05/21-13:09:04.744-0x7fa5c5a7c0-INFO-Available FALSE, retry... nLoop: 1
2025/05/21-13:09:04.795-0x7fa5c5a7c0-INFO-Available FALSE, retry... nLoop: 2
2025/05/21-13:09:04.845-0x7fa5c5a7c0-INFO-Validation Path to oexserverd is: /usr/lib/avnav/server/../plugins/ocharts/bin/oexserverd
2025/05/21-13:09:04.847-0x7fa5c5a7c0-INFO-Validation Path to server test is: /usr/lib/avnav/server/../plugins/ocharts/bin/oexserverd
2025/05/21-13:09:04.847-0x7fa5c5a7c0-INFO-o_charts_pi: Checking oexserverd utility at {/usr/lib/avnav/server/../plugins/ocharts/bin/oexserverd}
2025/05/21-13:09:04.847-0x7fa5c5a7c0-INFO-o_charts_pi: starting oexserverd utility: /usr/lib/avnav/server/../plugins/ocharts/bin/oexserverd
2025/05/21-13:09:05.879-0x7fa5c5a7c0-INFO-o_charts_pi: oexserverd Check OK...LoopCount: 4
2025/05/21-13:09:05.881-0x7fa5c5a7c0-INFO-verifySENC s0:  1  48399  0  0
2025/05/21-13:09:05.885-0x7fa5c5a7c0-INFO-   Cannot load SENC file /home/pi/avnav/data/ocharts/charts/__tmp2025-05-21-13-09-04-667222/OC-44-571003.oesu
2025/05/21-13:09:05.885-0x7fa5c5a7c0-INFO-Retry...
2025/05/21-13:09:05.887-0x7fa5c5a7c0-INFO-verifySENC s0:  1  48399  0  0
2025/05/21-13:09:05.891-0x7fa5c5a7c0-INFO-   Again, cannot load SENC file /home/pi/avnav/data/ocharts/charts/__tmp2025-05-21-13-09-04-667222/OC-44-571003.oesu
2025/05/21-13:09:05.891-0x7fa5c5a7c0-ERROR-opening /home/pi/avnav/data/ocharts/charts/__tmp2025-05-21-13-09-04-667222/OC-44-571003.oesu failed with error 10
2025/05/21-13:09:05.891-0x7fa5c5a7c0-ERROR-killing all children
2025/05/21-13:09:06.936-0x7fa5c5a7c0-INFO-Available FALSE, retry... nLoop: 1
2025/05/21-13:09:06.986-0x7fa5c5a7c0-INFO-Available FALSE, retry... nLoop: 2
2025/05/21-13:09:07.037-0x7fa5c5a7c0-INFO-Validation Path to oexserverd is: /usr/lib/avnav/server/../plugins/ocharts/bin/oexserverd
2025/05/21-13:09:07.037-0x7fa5c5a7c0-INFO-Validation Path to server test is: /usr/lib/avnav/server/../plugins/ocharts/bin/oexserverd
2025/05/21-13:09:07.037-0x7fa5c5a7c0-INFO-o_charts_pi: Checking oexserverd utility at {/usr/lib/avnav/server/../plugins/ocharts/bin/oexserverd}
2025/05/21-13:09:07.037-0x7fa5c5a7c0-INFO-o_charts_pi: starting oexserverd utility: /usr/lib/avnav/server/../plugins/ocharts/bin/oexserverd
2025/05/21-13:09:08.064-0x7fa5c5a7c0-INFO-o_charts_pi: oexserverd Check OK...LoopCount: 4
2025/05/21-13:09:08.065-0x7fa5c5a7c0-INFO-verifySENC s0:  1  48399  0  0
2025/05/21-13:09:08.069-0x7fa5c5a7c0-INFO-   Cannot load SENC file /home/pi/avnav/data/ocharts/charts/__tmp2025-05-21-13-09-04-667222/OC-44-571003.oesu
2025/05/21-13:09:08.069-0x7fa5c5a7c0-INFO-Retry...
2025/05/21-13:09:08.072-0x7fa5c5a7c0-INFO-verifySENC s0:  1  48399  0  0
2025/05/21-13:09:08.076-0x7fa5c5a7c0-INFO-   Again, cannot load SENC file /home/pi/avnav/data/ocharts/charts/__tmp2025-05-21-13-09-04-667222/OC-44-571003.oesu
2025/05/21-13:09:08.076-0x7fa5c5a7c0-ERROR-opening /home/pi/avnav/data/ocharts/charts/__tmp2025-05-21-13-09-04-667222/OC-44-571003.oesu finally failed after retry
2025/05/21-13:09:08.076-0x7fa5c5a7c0-INFO-ChartManager: TryOpenChart /home/pi/avnav/data/ocharts/charts/__tmp2025-05-21-13-09-04-667222/OC-44-571003.oesu returns 10
2025/05/21-13:09:08.076-0x7fa5c5a7c0-INFO-TryMessage done
2025/05/21-13:09:08.077-0x7f69bf6c00-INFO-Try open chart requestRequest: finished`

This is on a Raspberry pi 4

[image]

What is going wrong?

wellenvogel, 2025-05-21:
Seems that the key is not valid any more.
Are you sure that you used the correct system to assign your charts?
What er the names of the old chart directories? Same system name?

Fietsenwiel, 2025-05-21:
How can I check that for sure?
I have a dongle installed

wellenvogel, 2025-05-21:
But you use a chart set that was assigned to RelexedPi without the dongle (depends on what you do in the shop).
Can you show/send me the list of assignments in the shop?

Fietsenwiel, 2025-05-21:
Ok, I now learned that the dongle fr is starting with oc03x.
So I have the wrong fingerprint.
I'll trie again

Fietsenwiel, 2025-05-21:
Yes, it was the wrong fingerprintfile. Assumption is the mother of the devil.
The supplier of O-charts told me what the right fingerprint was (very quick response of them!)

I expected a more clear error message so i didn't think this was the problem.
Now it works.

Thanks for your support (again)!

closed 2025-05-21 by Fietsenwiel