wellenvogel/avnav #424: Error widget wind AVNav Version 20250602

niortais, 2025-06-07:
I use avnav  Version 20250602 with a yachta 2 anemometer. The widget displays the direction but not the speed.

[image]
On the other hand, page 2 of the instrument displays the speed and direction correctly.

[image]
The change of display between true wind or apparent wind is not taken.
I didn't have this problem with the previous version.

wellenvogel, 2025-06-08:
Would it be possible to provide some NMEA data for testing?
Which one was the previous version you have used?
And which system - Android or Windows or Linux?

niortais, 2025-06-09:
I use avnav on raspberry pi
The previous version is 20250606
The layout use is the the layout supplied as standard
The NMEA data go out from the anometer are (Capture made with socat on port 6666 of the anemometer which is a yachta V2)

$WIMWV,311.67,R,4.00,N,A*25
$WIVWR,48.33,L,4.00,N,2.06,M,7.41,K*77
$WIVPW,0.00,N,0.00,M*4c
$PWINF,0,311.67,D,0.02,D,2.06,M,7.41,K,4.00,N,1,B,*51
$PWWST,C,0,22.44,A*4b
$WIMWV,312.72,R,4.75,N,A*20
$WIVWR,47.28,L,4.75,N,2.44,M,8.79,K*72
$WIVPW,0.00,N,0.00,M*4c
$PWINF,0,312.72,D,0.02,D,2.44,M,8.79,K,4.75,N,1,B,*56
$PWWST,C,0,22.44,A*4b
$WIMWV,323.18,R,4.69,N,A*23
$WIVWR,36.82,L,4.69,N,2.41,M,8.68,K*7c
$WIVPW,0.00,N,0.00,M*4c
$PWINF,0,323.18,D,0.02,D,2.41,M,8.68,K,4.69,N,1,B,*50
$PWWST,C,0,22.44,A*4b
$WIMWV,324.52,R,4.53,N,A*23
$WIVWR,35.48,L,4.53,N,2.33,M,8.38,K*70
$WIVPW,0.00,N,0.00,M*4c
$PWINF,0,324.52,D,0.02,D,2.33,M,8.38,K,4.53,N,1,B,*50
$PWWST,C,0,22.44,A*4b
$WIMWV,323.47,R,4.27,N,A*23
$WIVWR,36.53,L,4.27,N,2.20,M,7.91,K*74
$WIVPW,0.00,N,0.00,M*4c
$PWINF,0,323.47,D,0.02,D,2.20,M,7.91,K,4.27,N,1,B,*5e
$PWWST,C,0,22.44,A*4b
$WIMWV,323.78,R,4.45,N,A*2b
$WIVWR,36.22,L,4.45,N,2.29,M,8.24,K*7e
$WIVPW,0.00,N,0.00,M*4c
$PWINF,0,323.78,D,0.02,D,2.29,M,8.24,K,4.45,N,1,B,*5e
$PWWST,C,0,22.44,A*4b
$WIMWV,323.75,R,5.30,N,A*25
$WIVWR,36.25,L,5.30,N,2.73,M,9.81,K*7b
$WIVPW,0.00,N,0.00,M*4c
$PWINF,0,323.75,D,0.02,D,2.73,M,9.81,K,5.30,N,2,B,*52
$PWWST,C,0,22.44,A*4b
$WIMWV,308.33,R,9.93,N,A*2b
$WIVWR,51.67,L,9.93,N,5.11,M,18.39,K*49
$WIVPW,0.00,N,0.00,M*4c
$PWINF,0,308.33,D,0.02,D,5.11,M,18.39,K,9.93,N,3,B,*6d
$PWWST,C,0,22.44,A*4b
$WIMWV,303.41,R,9.93,N,A*25
$WIVWR,56.59,L,9.93,N,5.11,M,18.39,K*43
$WIVPW,0.00,N,0.00,M*4c
$PWINF,0,303.41,D,0.02,D,5.11,M,18.39,K,9.93,N,3,B,*63
$PWWST,C,0,22.44,A*4b
$WIMWV,299.98,R,9.83,N,A*22
$WIVWR,60.02,L,9.83,N,5.06,M,18.21,K*46
$WIVPW,0.00,N,0.00,M*4c
$PWINF,0,299.98,D,0.02,D,5.06,M,18.21,K,9.83,N,3,B,*6b
$PWWST,C,0,22.44,A*4b
$WIMWV,300.11,R,9.83,N,A*22
$WIVWR,59.89,L,9.83,N,5.06,M,18.21,K*4f
$WIVPW,0.00,N,0.00,M*4c
$PWINF,0,300.11,D,0.02,D,5.06,M,18.21,K,9.83,N,3,B,*6b
$PWWST,C,0,22.44,A*4b
$WIMWV,299.89,R,9.83,N,A*22
$WIVWR,60.11,L,9.83,N,5.06,M,18.21,K*44
$WIVPW,0.00,N,0.00,M*4c
$PWINF,0,299.89,D,0.02,D,5.06,M,18.21,K,9.83,N,3,B,*6b
$PWWST,C,0,22.44,A*4b
$WIMWV,299.69,R,0.00,N,A*2e
$WIVWR,60.31,L,0.00,N,0.00,M,0.00,K*7d
$WIVPW,0.00,N,0.00,M*4c
$PWINF,0,299.69,D,0.02,D,0.00,M,0.00,K,0.00,N,0,B,*5d
$PWWST,C,0,22.44,A*4b

wellenvogel, 2025-07-23:
release with [20250723](https://www.wellenvogel.de/software/avnav/downloads/release/20250723/?lang=en)

closed 2025-07-23 by wellenvogel