wellenvogel/avnav #427: Duplicate Overlays cannot be deleted

nmeyne, 2025-06-10:
Unable to delete duplicate overlays from main screen or from overlay button.  Delete dialog appears to work, but on returning to chart icon on main page or on clicking chart detail, duplicates remain. 

Clicking on feature list rows repeatedly results in recursive display?

[image]

[image]

chrome version 137.0.7151.68 and firefox 139, with avnav/avnav-raspi 20250608

wellenvogel, 2025-06-11:
can reproduce this. Seems to be related to a chart overlay in the defaults. But I'm still testing this.

wellenvogel, 2025-06-30:
Just use drag&drop in the fefault overlay dialog to place the overlays above/below the chart.