wellenvogel/avnav #432: AvNav overzoom response

David-Sharman, 2025-06-14:
In AvNav when you zoom in on an area there comes a point where it goes to a white screen. If one zooms in on the same area in OCPN (and with the same charts) then one can continue to zoom in indefinitely and an ‘overzoom’ watermark will appear on the screen as a warning that one is at a higher zoom level than the chart information can support in a valid manner. In this instance the OCPN behaviour is vastly preferable to the AvNav behaviour. There are instances where one might quite legitimately overzoom, for example during a Man Overboard (MOB) or during an anchorage or anchor watch in a tight location. This is a safety issue in my opinion and so should be addressed.

(The basis of this observation is when using Greek vector charts from O-charts. Exactly the same charts are presented differently by AvNav and OCPN.)

David-Sharman, 2025-07-02:
Exactly the same charts are presented differently by AvNav and OCPN. This suggests to me it is an AvNav issue, not an O-charts issue.

wellenvogel, 2025-07-03:
No.