wellenvogel/avnav #436: AvNav - permissible file names

David-Sharman, 2025-06-14:
When I loaded mbTiles files there was then a dialog box to say that the mbtiles name was inadmissible. The name I had used (below) was autogenerated by SAS/Sat2Chart and was acceptable to OCPN on my Win10 machine, but it seems not to AvNav on Android.

"Ionian CS2S.Navionics Marine Charts.Z10+Z13+Z16+Z18.mbtiles"

The filename above includes some characters that appear to be inadmissible in Android, specifically ? "spaces". 

Suggestion : the dialog box as a minimum should say what the illegal characters in a given filename are. Ideally it should propose an amended filename with a simple 'accept proposed' button. For example by replacing every illegal character with an acceptable legal character such ? underscore _ . 

Also - Recommendations:
1. Why the red text ?
2. The whole process is not exactly intuitive.
3. Some indication of progress would be great.
4. Accepting the same file names in AvNav as in OCPN would be fantastic.

See https://www.cruisersforum.com/forums/f121/avnav-navigation-in-the-browser-235253-20.html#post3988818 and onwards for background info.

wellenvogel, 2025-06-16:
Could you add some screen shots for the issue you have with red text?
I can reproduce the problem that AvNav will be unable to use a chart that has a "+" in it's file name.
But I cannot see any of the other issues.
For thr upload issues with large files:
You should try to separate things - 
(1) Bring your chart to the memory of the android device
(2) Upload from there to AvNav
This will help to prevent some of the issues. When you try to upload directly from google drive both steps are executed somehow in the background - the first one out of control for AvNav.
If you have a working local network you could also follow the proposal you got in the forum - i.e. activate the webserver, allow external access and upload from an external device (e.g. laptop) via the browser.
Progress indicator is also available...

wellenvogel, 2025-07-23:
release with [20250723](https://www.wellenvogel.de/software/avnav/downloads/release/20250723/?lang=en)

closed 2025-07-23 by wellenvogel