wellenvogel/avnav #437: AVnav image: operating system (OS) does not indicate support for Raspberry Pi 5

hiaseilert, 2025-06-17:
**Steps to reproduce**
Write Rasperry PI OS image "2025-05-13-raspios-bookworm-arm64.img" to SD-Card.

Boot Rasperry PI OS from SD-Kard

Download AvNav-Image from https://www.free-x.de/raspbian/bullseye/image_2024-09-03-avnav-arm64-lite.zip

Use the Rasperry Pi imager and transfer it to the SSD (atached via m.2 HAT to the Rasperry) as described at http://www.raspberrypi.org/downloads (raw images).

Shot down the Pi, remove the SD

Power the pi up

**seen behavior**
Raspberry Pi 5 - 8GB
bootloader: 26826259 2024-09-23

(...)

The installed operating system (OS) does not indicate support for Raspberry Pi 5
Update the OS or set os_check=0 in config.txt to skip this check.
nvme: error 8
Failed to open device: 'nvme'
Boot mode: USB-MSD (04) order f

Expected behavior
The AVNav image boots

free-x, 2025-06-17:
Hi @hiaseilert ,

Images based on "bullseyes" don't support Pi5. You need image based on "bookworm"
We prepared "experimental" image based on it.
More Informations you can find at: https://github.com/free-x/AvNav-Image/blob/main/README.md

greetings
free-x

hiaseilert, 2025-06-18:
oh sorry, many thanks. we have only looked at the web site but not on github. We haven`d tried the bookworm image yet, but feel free to close the Issue. 
Viele Grüße, hias