wellenvogel/avnav #443: The FeatureList is shown while panning the chart

gertwin, 2025-07-03:
I have difficulties panning a chart on a touch screen which is connected to a Rasperry Pi.

System:
Raspberry Pi 5
AvNav image: 2025-03-19-avnav-bookworm-arm64-lite updated to version 20250702
Touch screen directly connected to the Raspberry Pi

Current behavior:
When I pan in the chart, every now and then the FeatureList window is shown.
This is inconvenient because I have to click away the FeatureList window and start panning again.

Expected behavior:
Panning the chart without getting the FeatureList window in the way.

A setting to disable the feature info on map clicks would be useful.

Thanks in advance.

wellenvogel, 2025-07-04:
Already addressed with daily [20250702](https://www.wellenvogel.net/software/avnav/downloads/daily/20250702/).

gertwin, 2025-07-04:
Yes, I can confirm it is working as expected.

closed 2025-07-04 by gertwin