wellenvogel/avnav #490: Feature request: user app: provide automatic reload option

W-Geronius, 2025-07-25:
This may be an improvement mainly for system testing & configuring.

This layout:

[image]

The right window is a user app defined by 

[image]

It would be very useful to provide an option to have this link refresh automatically. 
At present it just displays data fetched when the user app was opened.

wellenvogel, 2025-07-25:
I don't think that this should be implemented in AvNav.
It's very uncommon to use a webpage like this.
But you can easily create a HTML page in AvNav with some simple js code that fetches this data periodically.
Could look like
```
<html>
<head>
<style>
  p#data{
  white-space: pre;
}
</style>
<script type="text/javascript">
document.addEventListener("DOMContentLoaded", (event)=>{
  const el=document.getElementById('data');
  window.setInterval(()=>{
    fetch('/viewer/avnav_navi.php?request=gps')
    .then((resp)=>resp.json())
    .then((json)=>{
      el.textContent=JSON.stringify(json,undefined,2);
    })
  },2000)
})
</script>
</head>
<body>
<p id="data">Loading data</p>
</body>
</html>
```
Unfortunately currently you cannot add query parameters to internal urls.
But if you use this like an external URL you could also add query parameters and replace
```
fetch('/viewer/avnav_navi.php?request=gps')
by
fetch('/viewer/avnav_navi.php'+window.location.search)
```
You would set up the external URL like
`http://$HOST:8080/user/viewer/monitor.html?request=gps`
(assuming you named the page monitor.html)

And there is much more you could do with such a page.

wellenvogel, 2025-07-25:
Btw: You can use the browser developer tools to monitor network traffic easily...