wellenvogel/avnav #523: Satellites missing: Avnav NMEA pass through from GPS device to SocketWriter on port 34567

pozoo, 2025-08-24:
Hi,

I have a setup where avnav reads from a usb GPS device and gpsd connects to its SocketWriter on port 34567 and uses this to correct the system clock with chronyd. This works fine if gpsd is started after avnav. However I noticed that avnav reports many more satellites used for navigation (18) than gpsd (10) with the command "cgps -s". If I connect gpsd directly to the usb device, cgps also reports all satellites. The satellites missing are Beidou (BD) and some Galileo (GA). GPS (GP) and Glonass (GL) seem to be forwarded correctly. 

Therefore it seems avnav filters some of the NMEA0183 sentences out before writing them to the SocketWriter. Is this expected behavior?

pozoo, 2025-08-24:
Yes, plain Raspian bookworm + avnav package but without avnav-raspi package. I also have some plugins installed: ochartsng, history-plugin, mapproxy-plugin. There is not much else installed, it's a dedicated raspi for Avnav.

free-x, 2025-08-25:
>  Is there a way to provide the requested data outside GitHub? I would prefer not to post detailed GPS data here.

We do not have a standard procedure for this. It is your decision how you provide data.

> it's a dedicated raspi for Avnav.

So why not use ready-made images?

pozoo, 2025-08-25:
> So why not use ready-made images?

I setup this system 2 months ago. There was no stable bookworm image till last week and a pi5 does not work with pre bookworm images. Maybe I will give it a try it in the future.

closed 2025-10-04 by wellenvogel